pyrpl-fpga/pyrpl #479: InvalidStateError: Cannot take the result of a future that is never done

cici-hanna, 2022-08-04:
When running the oscilloscope code in section 3.2.3 of [this tutorial](https://pyrpl.readthedocs.io/en/latest/api.html), I get "InvalidStateError: Result is not set." when I reach line 57,
`ch1, ch2 = res.result()`

This error happens when you try to get the result of a future that isn't done, so its results are not yet available. According to [this documentation](https://pyrpl.readthedocs.io/en/latest/_modules/pyrpl/hardware_modules/scope.html), however, the future will never be done running:

>  This Future object is the one controlling the acquisition in
    rolling_mode. It will never be fullfilled (done), since rolling_mode
    is always continuous, but the timer/slot mechanism to control the
    rolling_mode acquisition is encapsulated in this object.

If I create a loop to end when the future stops running, to see if it's running, finishing, and then re-running in a loop, it never ends, so it seems like it just starts running the first time and never stops. So how can I ever get a result?

A-Rob-92, 2023-01-10:
Did you ever get to the bottom of this, I am having the same issue.

michaelcroquette, 2026-09-23:
This should be fixed with most recent version.

closed 2026-09-23 by michaelcroquette