pyrpl-fpga/pyrpl #491: Trouble installing PyRPL

AngeloManetta, 2023-01-19:
I am currently facing some trouble installing pyrpl through anaconda. I try to run 'pip install pyrpl' and I get this error message:

[image]

gsteele13, 2023-08-18:
I would recommend making a new environment and installing from fresh there, probably some complex dependency problem in your base env

For example:

```
conda create -n pyrpl3.8
conda activate pyrpl3.8
```
 
and then follow my instructions here for getting the right package versions:

https://github.com/lneuhaus/pyrpl/issues/468

Cheers,
Gary

michaelcroquette, 2025-10-28:
the code on pypi is outdated.

closed 2025-10-28 by michaelcroquette