pyrpl-fpga/pyrpl #492: Relative IQ Phase Change

DavidL-23, 2023-04-18:
Hello, 

I'm having issues when trying to change the relative phase between two signals. I am able to use the synchronize_iqs() function to synchronize the two signals to make sure they are starting at the same phase, which is something I do need, but when try to change the relative phase of one of the signals in the iq.setup( phase=?...) nothing actually happens. I can set it as any number but the two signals will always have a 0 relative phase. This also happens whether I use syncronize_iqs() or not.

I'm trying to change the relative phase of one of the signals because I'm trying to use a bucking coil to null the primary field of the my transmitter coil. Note that I'm using tune coils, as such when trying to null the primary field I need a specific amplitude and phase, as accurate to the hundredth decimal place, in order to accomplish getting a zero received signal. I have already accomplish this by using the Red Pitaya's own API/SCPI commands but I'm looking to use the Pyrpl library to have better/more control of the board. Wondering why I'm not able to change the phase of the signals when using Pyrpl but I am when using the Red Pitaya API.

SamuelDeleglise, 2023-04-18:
I am guessing we are not talking about the same phase... When you change the "phase" register of an IQ, you are changing the phase of the demodulation stage compared to that of the modulation (or vice-versa, I am not sure, one would have to check in the verilog code).
But it could well imagine that the signal you are looking at is the one that stays fixed, and not the one that is dephased with respect to it. Does that make sense ?

SamuelDeleglise, 2023-04-19:
One thing you can do is use one single IQ for both signals. The output_direct would go to one coil and then you set a non zéro quadrature factor and select "quadrature" for the output multiplexer. Then you only need to somehow route this output signal to an analog output of the board... This can be done with a PID, setting p=1 and output_direct= whatever you like