pyrpl-fpga/pyrpl #499: I need your help about unable recognizing red pitaya board when connecting with the computer.

hello-synD, 2023-08-03:
I planned to lock the Michelson interferometer by PID control using Red Pitaya Board STEM_125-14 FPGA. Although I set my laptop’s network setup as the video(https://www.youtube.com/watch?v=WnFkz1adhgs) explaining PyRPL program instructed and connected my laptop with router and the board by LAN cable, the program suddenly force-quitted. More detail, the program could not recognize my board, and I didn’t see board’s IP address and MAC address as well as port number. I tried other laptops running on Windows 7, 10, 11, Linux.
Last week, I found someone like me going through network connection issue and also a program Linien, for you to introduce.
Linien program recognize the Board, but still, there is a problem.

-First, I’d like to tell you what I went though on PyRPL.-
I connected my laptop, router, and red pitaya board by LAN cable. And then, I set my laptop’s network setup fixing IP address 192.168.1.101 as the video said. 
 I could enter the router’s setup web page, and check all of the IP address being connected or be allocated.
 So, I did ping test between my router and laptop, as well my board and laptop. There is no problem on ping test. Also, I could enter my board’s network web page and check IP address and port number, gateway and so on.
 But when I executed PyRPL.exe program, I could not select the IP address because that program might not recognize the board. Additionally, only “fake” item was still. I typed the board’s IP and port number and press OK button, and then PyRPL.exe program forcibly quitted. 
 I tried to approach by Anaconda manually, but still failed.
 According to my investigation, the borad had two different IP address, one was allocated by router, and the other was what the board originally had. 
 I tried to enter board’s web page again by the other IP address, and It did work equally.

 I installed apt install net-tools into the board because the board has a problem on network. And currently, I enter the board’s web page by the other IP.

-Second, trouble from Linien.-
This week, I installed the program Linien because I hoped to work properly. Although the program recognized the board’s IP(the other IP), I can’t see the signal on scope. For testing, I inputted the sinusoidal signal, but the scope display didn’t show anything. 

-Finally, about the program, Linien.-
Can Linien also use for locking of interferometer? Because, I saw a paper on the program in arxiv, yesterday. But the topic of the paper was for frequency stabilization of the laser when doing spectroscopic experiment.

hello-synD, 2023-08-03:
It is the similar to https://github.com/lneuhaus/pyrpl/issues/477

peteasa, 2024-11-01:
I have a potential fix for this problem see - https://github.com/RedPitaya/pyrpl/commit/ff7c04afca9114accb41d65e24be105db1d15da5