pyrpl-fpga/pyrpl #500: problem with async routines? (example in documentation fails!)

gsteele13, 2023-08-18:
Hi all, 

I must admit that although I've been using the RP for 4 years now, I've never figured out how to get the acquisition routines working properly (and have been using instead the hidden low-level `_start_acquisition` and `_get_curve`...). But I'd like to try to figure this out, since the others are giving me some strange results. 

What puzzles me is that `result()` for me never worked. But I've now figured it out: `result()` **only** works if it is **not** executed in the **same cell** as `curve_async()`!

This works fine:

[image]

But if I just merge the cells, it fails:

[image]

with error `Result not set`. 

Does anyone have any insight into this? 

Thanks!
Gary

gsteele13, 2023-08-18:
(i also confirmed that the same is true directly in the python interpreter, aka it's not a "notebook" problem...)

gsteele13, 2023-08-18:
Thanks for the tip!

I tried this: 

[image]

but as you see I got the same result. 

I also have never gotten `scope.single()` to work: it just hangs forever. I am also not able to interrupt it: if I run `scope.single()`, my whole notebook crashes for ever and I have to restart the kernel....

SamuelDeleglise, 2023-08-18:
I am not sure what's going wrong. I am on branch python3-only (which was just merged into master, so I expect that master should be identical), and I have the following:
[image]

closed 2025-10-23 by michaelcroquette