pyrpl-fpga/pyrpl #502: how to get I and Q of iq0 and iq1?

gsteele13, 2023-10-31:
I hope it's OK to post this here...

We've been using the on-board RP IQ modules a lot lately, and I've got an application in mind in which I would like to run simultaneously an IQ demodulator on both input1 and input2 (I want to build a finite IF GHz VNA and need one for the device path and one for the reference path)

I understand that pyrpl implements three iq demodulators (iq0, iq1, and iq2). However, as far as I can figure out, I can only get both the I and the Q data from iq2, eg. by connecting the scope input to streams iq2 and iq2_2 (which we interpret as the real an imaginary quadratures of iq2). 

Is there some way to get both IQ quadratures out of iq0 and iq1? 

And maybe also important: if I set say iq1 and iq2 to the same reference frequency, do they use the same digital clock reference for demodulation? aka. will their phase be locked? 

I'm trying to figure out if pyRPL will meet my application needs, or if I need to go to something like one of the SDR streamers

Thanks!
Gary

SamuelDeleglise, 2023-11-06:
Hi, sorry for late reply, the answer is unfortunately, you would have to modify the FPGA code in order to output the imaginary part Q for IQ0 or IQ1. At the moment, this is only possible with IQ2 as you mentionned. The good news is that it's probably not too hard to figure out how to do it based on what was done with IQ2 already.

Regarding the other part of your question: this is definitely doable to have the 3 IQ perfectly in phase, you should first set them to the same frequency, and then call the special command "synchronize_iqs" that will reset the phases of all the 3 IQ internal clocks at precisely the same point.

Good luck with your application

gsteele13, 2023-11-06:
Thanks! Ill have a look. Am also looking at the sdr streaming apps, they
might be exactly what we need

closed 2024-12-04 by michaelcroquette