pyrpl-fpga/pyrpl #503: the system will abruptly exit。

llzy665, 2023-11-11:
When I use the [precompiled executable for windows “pyrpl-windows.exe”],I cam connect my pitaya but when I reached the interface for selecting the hostname, I couldn't locate my designated hostname, causing the system to abruptly exit。
[image]

martin-santiago, 2023-12-05:
Hi @llzy665  , this happened to me, have you found any solution?, also it doesn't recognize my redpitaya and I've made sure (several times) that I'm connected on the same lan, in fact I can go to the default `rp-XXXX.local` but for some reason I'm not able to get it recognized by pyrpl

TaoChan1005, 2024-01-23:
I also met this problem with my new PC. Even when I designed the IP address, it still crashed.  I can run the .exe in my other laptop(13xxx Intel CPU) and my VM.  

My new PC uses the 14600K CPU, I am not sure if this is a problem.

Goldfish76, 2025-08-04:
Yeah experiencing exactly the same issue, anyone know how to fix it?

michaelcroquette, 2025-08-04:
What happens if you just enter the redpitaya ip address manually ? Can you acces the RP interface by typing the IP address in your browser ?

Goldfish76, 2025-08-04:
> What happens if you just enter the redpitaya ip address manually ? Can you acces the RP interface by typing the IP address in your browser ?

Here's the result when I manually enter the IP address on pyrpl, and by enter this to the browser I can access the RP interface:
D:\Downloads>pyrpl_04-08-2025
DEBUG:pyrpl.async_utils:Could not enable IPython gui support: 'NoneType' object has no attribute 'run_line_magic'.
DEBUG:pyrpl.attributes:SelectProperty unit of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty unit of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty unit of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty unit of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
0 pyrpl_04-08-2025
Calling Pyrpl(**{})
INFO:pyrpl:Please select or create a configuration file in the file selector window!
INFO:pyrpl.redpitaya:Please choose the hostname of your Red Pitaya in the hostname selector window!
INFO:pyrpl.redpitaya:FPGA is being flashed. Please wait for 2 seconds.
WARNING:pyrpl.redpitaya:no pyrpl instance
WARNING:pyrpl.redpitaya:no pyrpl instance
INFO:pyrpl.redpitaya:Successfully connected to Redpitaya with hostname 192.168.137.218.
Traceback (most recent call last):
  File "pyrpl\__main__.py", line 40, in <module>
  File "pyrpl\pyrpl.py", line 327, in __init__
  File "pyrpl\software_modules\network_analyzer.py", line 191, in _load_setup_attributes
  File "pyrpl\modules.py", line 514, in _load_setup_attributes
  File "pyrpl\modules.py", line 501, in setup_attributes
  File "pyrpl\modules.py", line 172, in setup
  File "pyrpl\software_modules\network_analyzer.py", line 631, in _setup
  File "pyrpl\acquisition_module.py", line 411, in _setup
  File "pyrpl\acquisition_module.py", line 385, in stop
  File "pyrpl\software_modules\network_analyzer.py", line 636, in _free_up_resources
  File "pyrpl\software_modules\network_analyzer.py", line 207, in iq
  File "pyrpl\attributes.py", line 240, in __set__
  File "pyrpl\attributes.py", line 724, in validate_and_normalize
  File "pyrpl\attributes.py", line 798, in valid_frequencies
  File "pyrpl\attributes.py", line 790, in _MAXSHIFT
ZeroDivisionError: float division by zero
[PYI-32508:ERROR] Failed to execute script '__main__' due to unhandled exception!

michaelcroquette, 2025-08-04:
ok this is a known bug that we are trying to fix. Can you try using a new config file ? I doubt this will work but just in case. You can try to downgrade the OS that you flash on the SD card to 1.04, 0.94 or 0.92 for now. I'm trying to see if I can replicate.

Goldfish76, 2025-08-04:
> ok I can replicate. If you can run the python code you should not have this issue. I'm trying to fix the issue with the exe.

Thanks a lot for you help! I downgrade the OS to 0.94 and I can connect to the exe. 
for the python code do you mean that one installed by pip install pyrpl?

michaelcroquette, 2026-01-06:
Are you using the latest releases (https://github.com/pyrpl-fpga/pyrpl/releases) ? Normally this issue is fixed.

michaelcroquette, 2026-01-06:
Ok so this means the fpga bit file is not loaded properly. Which board and OS are you using ? Can you set `loglevel: debug` in the config file and paste the output ?

khalidali63624-ship-it, 2026-01-06:
> 这就解释了原因，这个exe文件目前还不兼容第二代产品。脚本正在“/opt/redpitaya/fpga/z10_125_pro_v2/pyrpl/fpga.bit.bin”目录下查找pyrpl FPGA镜像文件，但由于RedPitaya团队尚未将其复制到该目录，因此找不到该文件。您可以检查SD卡上是否存在“/opt/redpitaya/fpga/z10_125/pyrpl”文件夹，如果存在，则将pyrpl文件夹复制到“/opt/redpitaya/fpga/z10_125_pro_v2”目录下。这样应该就可以了。

Yes, that's right! Your method is correct. Thank you very much!

closed 2026-01-06 by michaelcroquette