pyrpl-fpga/pyrpl #504: Circumventing the IQ module in the pdh input of the Fabry Perot lockbox module

JArnbak, 2023-11-15:
Hi, 

I am trying to implement at PDH lock in pyrpl using an error signal already down-mixed in the detector that detects the cavity reflectivity .
I would like to use the Fabry Perot lockbox module, but the PDH input into that module automatically couples to the IQ module because it expects an AC signal to downmix into the final error signal. Since I already have a final error signal, I do not need this. I have tried to circumvent this by setting the modulation frequency to 0 and the amplitude to 1 to configure the IQ module as a "all-pass" filter for my error signal. From what I can see of the source code, the input high-pass filter is automatically set to have a cut-off freuqency of freq_mod /128, meaning setting the modulation frequency to 0 should also "disable" this highpass filter. However this does not seem to work, and the signal I get out is tiny (~1e-6 Vpp) compared to the input (almost 900 mVpp).

Is there any way of cirucmventing the IQ module in the Fabry Perot, or am I doing something wrong in configuring the IQ?

JArnbak, 2023-11-15:
sorry I misunderstood how the IQ module is configured. The quadrature factor should be 1, the phase 90 and amplitude should be 0 in order to configure the module to be "all-pass". However, this still does not work

michaelcroquette, 2024-03-08:
You need to use the Analog PDH input. It is not present by default in the Faby-Perot lockbox but it is defined in the code so if you create your own lockbox you can use it. Let me know if you need help doing that.