pyrpl-fpga/pyrpl #507: Derivative term in the PID

Bentwin2002, 2024-01-29:
Do you know why the D term in the PID was deactivated? Has anyone had any success reactivating and using the D term?

michaelcroquette, 2024-03-08:
The D term was unused in our experiments most of the time so we removed it. It is active in the default RedPitaya PID module if you really need it.

mixgreen, 2024-06-21:
> The D term was unused in our experiments most of the time so we removed it. It is active in the default RedPitaya PID module if you really need it.

Is that mean i can't use the D term in pyrpl?  if i really need it , is that posiable to uncomment the code to use it?
[image]
[image]
[image]

michaelcroquette, 2024-06-21:
Actually you can ! I thought the derivative had been disabled in fpga code but it was actually uncommented in #22. There is a comment in the code saying  "Derivative - 2 cycles delay (but treat as 1 cycle because its not functional at the moment" but I'm not sure what it means. 
The python code of the PID module was commented in #299 but there is no mention of this modification in the commit description. 
Anyway you can try and see ! It looks like the fpga code is the same as the original one from redpitaya so I think it should work. If it works we will uncomment it.

michaelcroquette, 2025-02-06:
Hello Nicolas,

I will create a branch on the git with the derivative active and without the IIR module (I think the reason Leo removed the derivative term was to gain computation power to fit the (big) IIR module on the FPGA). If you already have a branch you can publish it and send it and I will compile it.

nicoleroy-ipgp, 2025-02-06:
Hello Michael,
Thanks a lot ! 
I managed to recompile the FPGA code with DERIVATIVE = 1 (using Vivado 2015.4) but it does not seem to work. The available range of "d" parameter is quite small (but it might be normal) and nothing changes when I modify it. I am not sure that the code has been downloaded to the Red Pitaya (I put it in /opt/pyrpl with the right name following red_pitaya.py code). 
I also tried to use this branch : https://github.com/RedPitaya/pyrpl/tree/with_differentiator_piD but this does not work (it seems quite old ...)
And no, I have no branch.
Thanks for your help !

michaelcroquette, 2025-02-06:
So I guess you have your local copy of the code you modified and compile ? There's no need to "put" the code anywhere else than where it was before. What is the folder /opt/pyrpl ? 

You can comment the IIR module generation part to check if you are actually using the proper code (check c1295696f0474a20213f770cc4463c84d8c6eaa5).

nicoleroy-ipgp, 2025-02-06:
Yes I modified both the python and FPGA codes locally to try a few modifications (I don't fully understand the whole code yet ...). Then I compiled the FPGA code and put the generated red_pitaya.bin file in the directory /opt/pyrpl with name fpga.bin whichare the defaultparameters in file redpitaya.py (I understood that this file is copied when pyrpl is started depending on the options in redpitaya.py). Otherwise I don't understand how the modified FPGA firmware is uploaded to the FPGA. But I am fairly new to Red Pitaya and FPGA so my understanding of the whole process is limited ...
Thanks for your help !

nicoleroy-ipgp, 2025-02-13:
I did a few tests and I have the confirmation that my FPGA code is not uploaded to the Red Pitaya. So I guess that I am missing something in the whole process ...
Once the file red_pitaya.bin is generated, is there anything to do to upload it to the device ?
Thanks !

michaelcroquette, 2025-02-13:
This could be the issue. If you look at the last Pull requests there are some suggestions to resolve compatibility issues with the new RedPitaya OS. I'm lacking time for now to review those PRs and to try and solve your issue.
Good luck !

michaelcroquette, 2025-02-17:
I created this [branch] (https://github.com/pyrpl-fpga/pyrpl/tree/derivative_on_PID) where I've modified the FPGA code so the derivative is active. There still seems to be saturation issues but you can already test it if you want.

nicoleroy-ipgp, 2025-02-18:
Thanks a lot ! I managed to compile the FPGA code but I am still unable to upload it  to the FPGA. I tried OS 0.94 and OS2.00-37 with no luck ... I get the following error : 
sh: 1: echo: echo: I/O error
BIN FILE loading through FPGA manager failed

I tried to generated the bin file using bootgen etc... but nothing seems to work. 

Could you share what version of Red Pitaya OS you are using for your tests, please ?

Thanks a lot for your help !

michaelcroquette, 2025-02-18:
I'm using this [OS](https://sourceforge.net/projects/pyrpl/files/SD_Card_RedPitayaOS_v0.92.img.zip/download). Once you've compiled there is nothing to do apart from restarting pyrpl with `reload_fpga=True` and it will load the new generated bitfile. But yes the loading of the bitfile is different on the new OS, you can try Pete's [solution](https://github.com/pyrpl-fpga/pyrpl/pull/547) if you want to work with a newer OS but I haven't tested it.

nicoleroy-ipgp, 2025-02-19:
The issue was indeed the version of the FPGA. I had to use` "set part xc7z020clg400-1"` in red_pitaya_vivado.tcl file instead of `"set part xc7z010clg400-1"`.
I managed to compile FPGA code but I had to use bootgen to generate the bin file from the bit file because I got an error using the make script at the end (memory size) : 
```
echo -n "all:{ red_pitaya.bit }" >  red_pitaya.bif
bootgen -image red_pitaya.bif -arch zynq -process_bitstream bin -o red_pitaya.bit.bin -w
```
 Then I uploaded it manually to the FPGA using `fpgautil -b red_pitaya.bit.bin `

Now I have a derivative component for the PID but it seems strange. I will dig in the PID code to understand how it works ...

Thanks for your help !

michaelcroquette, 2025-02-19:
Good to know that you can compile the FPGA code for the z7020 version like that. Might be worth it to run the tests to see if everything is behaving correctly. Regarding the derivative, I saw some really weir behaviour so there is something wrong but I wasn't able to point it exactly. Good luck !

peteasa, 2025-03-03:
The other option is to use the pyrpl bit.bin file from the RedPitaya OS2 system..

peteasa, 2025-03-03:
> I'm using this [OS](https://sourceforge.net/projects/pyrpl/files/SD_Card_RedPitayaOS_v0.92.img.zip/download). Once you've compiled there is nothing to do apart from restarting pyrpl with [...]

Note that my solution should work with both the OS2 system and the older system from https://sourceforge.net/projects/pyrpl/files/

My code detects that you are using an older RedPitaya OS and behaves in a legacy way.. [547](https://github.com/pyrpl-fpga/pyrpl/pull/547) has been designed explicitly to support the older and the new OS..  backward compatible and all that!