pyrpl-fpga/pyrpl #509: 0 Hz beatnote/non-offset phase lock? 2 input CORDIC request

EsbenMadsen, 2024-03-25:
I am an electrical engineer writing my master thesis, so i apologize for any misunderstandings in the following:

As far as i can tell the only phase lock implemented in PYRPL is the offset phase lock, where a frequency difference in light generates a beat note on a photo detector due to the periodic change in relative phase (which i feel stretches the definition of a phase lock, but i digress) and this beat note is then demodulated into IQ components. For my setup the beat note should ideally be DC/0 Hz, so demodulating at any frequency is not an option. Instead i have two balanced photo detectors measuring the output of a 2x4 MMI, corresponding to the I and Q of the relative phase of the input lasers. From this i want to calculate the angle of the resulting phasor in order separate phase noise from power noise by using arctan(Q/I), which i can see from the documentation should be possible through the CORDIC algorithm. So here's my problem:

The IQ modules only allow you to apply CORDIC to one input signal and a local oscillator, whereas i want to apply CORDIC to 2 input signals (I and Q) and extract the phase information for PID control. Is this possible?

Todaycrrc, 2024-05-07:
hi guys Do you have any ideas yet?

EsbenMadsen, 2024-05-13:
> hi guys Do you have any ideas yet?

I tried looking into the FPGA firmware to identify if the CORDIC algorithm is implemented as a separate module that takes two inputs, but i didn't find anything i could use with my limited time. If the CORDIC implementation does exist as a module, then it should be easy to change both inputs to the module to be the analog inputs.

michaelcroquette, 2024-06-21:
Hello,

Currently the cordic algorithm is only implemented as a part of the IQ module and it's not possible to perform it using two separates I and Q inputs. This would require some modification of the FPGA code. Can't you use only the I signal as your error signal ? I'm not familiar with MMI.

EsbenMadsen, 2024-06-24:
Hi Michael, thank you for answering and for your part in creating PYRPL. 

My project has turned another direction so i haven't explored the issue further, but i believe it should be possible with relatively minor changes to the FPGA code. An MMI is an integrated version of a 90 degree hybrid, which decomposes light into I and Q phase signals using interference in a photonic chip. This is important when you want to measure the phase of light separately from its intensity. By calculating the phasor from the I and Q signals, you get phase as the angle (arctan[I/Q]) and intensity/power as the magnitude. PYRPL is an ideal candidate for implementing this calculation fast and precisely since it can also plot the phase/magnitude and even route the signals to a PID controller in a all-in-one commercial FPGA. So for instance if i were to use the phase of light in a control loop, I'd be able to separate phase changes from intensity fluctuations very easily.

NettleTea, 2024-11-14:
I have implemented 0-offset phase-locking of lasers using Pyrpl and Red Pitaya using a dither optical phase-locked loop. You can read more about it here [[10.1109/JLT.2023.3330707](https://doi.org/10.1109/JLT.2023.3330707)](url). The dither-loop is neat if you need a 0 Hz offset between the lasers. Also, you should be able to implement the lock without phase-modulators if you prefer, by using proportional gain of the laser current (for phase-control) and also outputting the IQ direct out onto the laser current (as the dither). Also make sure to syncronize the loop with your dither phase:
[image]

If you are using a phase-modulator to implement the phase-lock, have a look at issue #484 and the 2pi-phase jump solution mentioned there.