pyrpl-fpga/pyrpl #512: AttributeError: 'NoneType' object has no attribute 'endswith'

edofazza, 2024-05-06:
I was running this code 

`
from pyrpl import RedPitaya

class RedPitayaController():
    def __init__(self, hostname: str, user: str = 'root', password: str = 'root', config: str = 'fermi',
                 gui: bool = False):
        try:
            self.redpitaya = RedPitaya(hostname=hostname, config=config, user=user, password=password)
        except Exception as e:
            print(e)

    def reset(self) -> None:
        # Turn off arbitrary signal generator channel 0
        self.set_asg0(output_direct='off', amp=0, offset=0)
        # Turn off arbitrary signal generator channel 1
        self.set_asg1(output_direct='off', amp=0, offset=0)
        # Turn off I+Q quadrature demodulation/modulation modules
        self.redpitaya.iq0.output = 'off'
        self.set_iq0(output_direct='off')
        # Turn off PID module 0
        self.set_pid0(0, 0, 0, 0, 'off')
        # Turn off dac2
        self.set_dac2(0)

    def set_asg0(self, waveform: str = 'halframp', output_direct: str = 'out1', amp: float = 0.5,
                 offset: float = 0.5, freq: float = 1e2) -> None:
        self.redpitaya.asg0.setup(waveform=waveform, output_direct=output_direct, trigger_source='immediately',
                                  offset=offset, amplitude=amp, frequency=freq)

    def set_asg1(self, waveform: str = 'halframp', output_direct: str = 'out1', amp: float = 0.5,
                 offset: float = 0.5, freq: float = 1e2) -> None:
        self.redpitaya.asg1.setup(waveform=waveform, output_direct=output_direct, trigger_source='immediately',
                                  offset=offset, amplitude=amp, frequency=freq)

    def set_iq0(self, frequency: float = 25e6, bandwidth: list = [2e6, 2e6], gain: float = 0.5, phase: int = 0,
                acbandwidth: float = 5e6, amplitude: float = 1., input: str = 'in1', output_direct: str = 'out2',
                output_signal: str = 'quadrature', quadrature_factor: int = 1) -> None:
        self.redpitaya.iq0.setup(frequency=frequency, bandwidth=bandwidth, gain=gain, phase=phase,
                                 acbandwidth=acbandwidth, amplitude=amplitude, input=input, output_direct=output_direct,
                                 output_signal=output_signal, quadrature_factor=quadrature_factor)

    def set_dac2(self, voltage: float = 0.) -> None:
        self.redpitaya.ams.dac2 = voltage   # pin 17 output 0

    def set_pid0(self, ival: float = 0, integrator: float = 1e3, proportional: float = 0,
                 differantiator: float = 0, input='iq0', output_direct: str = 'out1') -> None:
        # Clear integrator
        self.redpitaya.pid0.ival = ival
        # Proportinal
        self.redpitaya.pid0.p = proportional
        # Integrator
        self.redpitaya.pid0.i = integrator
        # differentiator
        self.redpitaya.pid0.d = differantiator
        # input or output
        self.redpitaya.pid0.input = input
        self.redpitaya.pid0.output_direct = output_direct

p = RedPitayaController(hostname='169.254.167.128')
p.reset()
`

and I ended up with the following error:

INFO:pyrpl.pyrpl.redpitaya:Successfully connected to Redpitaya with hostname 169.254.167.128.
Traceback (most recent call last):
  File "rpcontrol.py", line 61, in <module>
    p.reset()
  File "rpcontrol.py", line 19, in reset
    self.set_iq0(output_direct='off')
  File "rpcontrol.py", line 40, in set_iq0
    output_signal=output_signal, quadrature_factor=quadrature_factor)
  File "/root/miniconda3/envs/redpitaya/lib/python3.7/site-packages/pyrpl/pyrpl/modules.py", line 165, in setup
    setattr(self, key, value)
  File "/root/miniconda3/envs/redpitaya/lib/python3.7/site-packages/pyrpl/pyrpl/attributes.py", line 240, in __set__
    value = self.validate_and_normalize(obj, value)
  File "/root/miniconda3/envs/redpitaya/lib/python3.7/site-packages/pyrpl/pyrpl/hardware_modules/dsp.py", line 111, in validate_and_normalize
    options = [o for o in self.options(obj) if o.endswith(value)]
  File "/root/miniconda3/envs/redpitaya/lib/python3.7/site-packages/pyrpl/pyrpl/hardware_modules/dsp.py", line 111, in <listcomp>
    options = [o for o in self.options(obj) if o.endswith(value)]
AttributeError: 'NoneType' object has no attribute 'endswith'

jl0up, 2024-06-28:
Does it work if you initialize like this instead ?

```python
from pyrpl import Pyrpl
self.pyrpl = Pyrpl(hostname=hostname, config='', gui=False)
self.redpitaya = self.pyrpl.rp
```

sankethan, 2024-10-22:
Any updates on this?

michaelcroquette, 2024-10-22:
Are you experiencing the same issue ? If so, can you provide us a code that I can quickly test to recreate the error.

michaelcroquette, 2024-10-22:
This code works for me on a simulated RedPitaya

```
from pyrpl import RedPitaya

class RedPitayaController():
    def __init__(self, hostname: str, user: str = 'root', password: str = 'root', config: str = 'fermi', gui: bool = False):
        try:
            self.redpitaya = RedPitaya(hostname=hostname, config=config, user=user, password=password)
        except Exception as e:
            print(e)

    def reset(self) -> None:
        # Turn off arbitrary signal generator channel 0
        self.set_asg0(output_direct='off', amp=0, offset=0)
        # Turn off arbitrary signal generator channel 1
        self.set_asg1(output_direct='off', amp=0, offset=0)
        # Turn off I+Q quadrature demodulation/modulation modules
        self.redpitaya.iq0.output = 'off'
        self.set_iq0(output_direct='off')
        # Turn off PID module 0
        self.set_pid0(0, 0, 0, 0, 'off')
        # Turn off dac2
        self.set_dac2(0)
    
    def set_asg0(self, waveform: str = 'halframp', output_direct: str = 'out1', amp: float = 0.5,
                 offset: float = 0.5, freq: float = 1e2) -> None:
        self.redpitaya.asg0.setup(waveform=waveform, output_direct=output_direct, trigger_source='immediately',
                                  offset=offset, amplitude=amp, frequency=freq)
    
    def set_asg1(self, waveform: str = 'halframp', output_direct: str = 'out1', amp: float = 0.5,
                 offset: float = 0.5, freq: float = 1e2) -> None:
        self.redpitaya.asg1.setup(waveform=waveform, output_direct=output_direct, trigger_source='immediately',
                                  offset=offset, amplitude=amp, frequency=freq)
    
    def set_iq0(self, frequency: float = 25e6, bandwidth: list = [2e6, 2e6], gain: float = 0.5, phase: int = 0,
                acbandwidth: float = 5e6, amplitude: float = 1., input: str = 'in1', output_direct: str = 'out2',
                output_signal: str = 'quadrature', quadrature_factor: int = 1) -> None:
        self.redpitaya.iq0.setup(frequency=frequency, bandwidth=bandwidth, gain=gain, phase=phase,
                                 acbandwidth=acbandwidth, amplitude=amplitude, input=input, output_direct=output_direct,
                                 output_signal=output_signal, quadrature_factor=quadrature_factor)
    
    def set_dac2(self, voltage: float = 0.) -> None:
        self.redpitaya.ams.dac2 = voltage   # pin 17 output 0
    
    def set_pid0(self, ival: float = 0, integrator: float = 1e3, proportional: float = 0,
                 differantiator: float = 0, input='iq0', output_direct: str = 'out1') -> None:
        # Clear integrator
        self.redpitaya.pid0.ival = ival
        # Proportinal
        self.redpitaya.pid0.p = proportional
        # Integrator
        self.redpitaya.pid0.i = integrator
        # differentiator
        self.redpitaya.pid0.d = differantiator
        # input or output
        self.redpitaya.pid0.input = input
        self.redpitaya.pid0.output_direct = output_direct
p = RedPitayaController(hostname='_FAKE_')
p.reset()
```

sankethan, 2024-11-01:
I solved this issue by initializing redpitaya this way.
```
p = pyrpl.Pyrpl(config=CONFIG, hostname=HOSTNAME, gui=False)
r = p.rp
```
**rather than**
```
r = RedPitaya(hostname=HOSTNAME)
```