pyrpl-fpga/pyrpl #513: No output from asg command

shivarajkandhasamy, 2024-05-20:
I am working with a Debian 11 system and have installed pyrpl (0.9.6.0)from GitHub master in a conda environment (python 3.9). The setup is able to recognize the Redpitya board (running with their 2.0 OS) and I can also see the inputs to the board through pyrpl's  'scope' module. However the 'asg' module does not work, it doesn't give any output when requested. Below is the sequence of steps I am running for this.

` 
(pyrpl-env) controls@cdscontrol:~$ python3
Python 3.9.19 (main, May  6 2024, 19:43:03)
[GCC 11.2.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pyrpl import Pyrpl
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
DEBUG:pyrpl.async_utils:Could not enable IPython gui support: 'NoneType' object has no attribute 'magic'.
/home/controls/anaconda3/envs/pyrpl-env/lib/python3.9/site-packages/paramiko/transport.py:219: CryptographyDeprecationWarning: Blowfish has been deprecated and will be removed in a future release
  "class": algorithms.Blowfish,
DEBUG:pyrpl.attributes:SelectProperty unit of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty unit of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty unit of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty unit of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
DEBUG:pyrpl.attributes:SelectProperty default_sweep_output of module None has no options!
>>> HOSTNAME = "192.168.19.10"
>>> p = Pyrpl(config='',hostname=HOSTNAME, gui=False)
INFO:pyrpl.redpitaya:Successfully connected to Redpitaya with hostname 192.168.19.10.
WARNING:pyrpl.modules:Register input of module pwm0 has value 0, which does not correspond to selected option 15. Setting to 'off'.
>>> r=p.rp
>>> asg = r.asg0
>>> asg.setup(frequency=1, amplitude=0.8, waveform='sin', trigger_source='immediately', output_direct='out1')
>>>
` 
After the above set of commands, I do not see any output from the 'out1' port of the Red Pitay board. Not sure if the warnings I see above about some of the attributes is indicating something. 
FYI, if I use the Red Pitya web app I can generate output from the 'out1' port, so I don't think it is a hardware problem.  Please let me know if there is a way to debug this issue. Thanks.

michaelcroquette, 2024-06-21:
Not sure what is happening but the warning message suggests that the lockbox module is trying to set up some modules. So you might have a conflict between two modules outputting signal on out1. I suggest you use the scope to debug. You can visualize the signal from asg0 and out1 separately and see if they match. Good luck !

shivarajkandhasamy, 2024-10-25:
I don't see this problem with the new versions of Redpitya OS.