pyrpl-fpga/pyrpl #514: Can't install pyrpl with pip

felixhoff, 2024-05-28:
If I run the pip installation, I get this syntax error : 

```
(base) C:\Users\PC>pip install pyrpl
Collecting pyrpl
  Using cached pyrpl-0.9.3.6-py2.py3-none-any.whl.metadata (9.0 kB)
Collecting futures (from pyrpl)
  Using cached futures-3.0.5.tar.gz (25 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [30 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 14, in <module>
        File "C:\Users\PC\anaconda3\Lib\site-packages\setuptools\__init__.py", line 266, in <module>
          monkey.patch_all()
        File "C:\Users\PC\anaconda3\Lib\site-packages\setuptools\monkey.py", line 98, in patch_all
          patch_for_msvc_specialized_compiler()
        File "C:\Users\PC\anaconda3\Lib\site-packages\setuptools\monkey.py", line 165, in patch_for_msvc_specialized_compiler
          patch_func(*msvc14('_get_vc_env'))
                      ^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\PC\anaconda3\Lib\site-packages\setuptools\monkey.py", line 155, in patch_params
          mod = import_module(mod_name)
                ^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\PC\anaconda3\Lib\importlib\__init__.py", line 126, in import_module
          return _bootstrap._gcd_import(name[level:], package, level)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\PC\anaconda3\Lib\site-packages\setuptools\_distutils\_msvccompiler.py", line 20, in <module>
          import unittest.mock as mock
        File "C:\Users\PC\anaconda3\Lib\unittest\mock.py", line 26, in <module>
          import asyncio
        File "C:\Users\PC\anaconda3\Lib\asyncio\__init__.py", line 8, in <module>
          from .base_events import *
        File "C:\Users\PC\anaconda3\Lib\asyncio\base_events.py", line 18, in <module>
          import concurrent.futures
        File "C:\Users\PC\AppData\Local\Temp\pip-install-g5ihizt7\futures_21e751383dc344969ba22a0e80de4658\concurrent\futures\__init__.py", line 8, in <module>
          from concurrent.futures._base import (FIRST_COMPLETED,
        File "C:\Users\PC\AppData\Local\Temp\pip-install-g5ihizt7\futures_21e751383dc344969ba22a0e80de4658\concurrent\futures\_base.py", line 357
          raise type(self._exception), self._exception, self._traceback
                                     ^
      SyntaxError: invalid syntax
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

shshae, 2024-06-07:
This happened with me when I was accidentally using python 3.8. I changed to python 3.6 and it worked.

michaelcroquette, 2024-06-21:
The pyrpl version on PyPI is quite outdated so I would advise to clone the repository from github if you want a version compatible with python 3.9.

jl0up, 2024-06-28:
Commenting here because I you may want to update setup.py. : I had to when I installed PyRPL in my conda environment from source using the method below. Is there a better method to install in development mode ?

First I cloned and created a conda environment:
```shell
cd c:\user\labo\documents\github
git clone https://github.com/lneuhaus/pyrpl.git
cd pyrpl
git checkout python3-only
conda env create -n pyrpl -f environment_pyrpl.yml
conda activate pyrpl
```

Then tried `python setup.py develop` and I got this error, along with some setuptools related deprecation warnings
`error: Couldn't find a setup script in C:\Users\labo\AppData\Local\Temp\easy_install-zhbtau4a\PyQt5-5.14.0.tar.gz`

**It worked after I removed the `<=5.14` requirement on PyQt5.** (version is 5.15.10 in `requirement_pyrpl.yml`

PS:
I also tried `pip install -e .`, it worked but it installed many pip packages that don't even appear in setup.py:
```
attrs-23.2.0
beautifulsoup4-4.12.3 
bleach-6.1.0 
defusedxml-0.7.1 
fastjsonschema-2.20.0 
importlib-metadata-8.0.0 
ipython-genutils-0.2.0 
jinja2-3.1.4 
jsonschema-4.22.0 
jsonschema-specifications-2023.12.1 
jupyter-core-5.7.2 
jupyterlab-pygments-0.3.0 
markupsafe-2.1.5 
mistune-3.0.2 
nbclient-0.10.0 
nbformat-5.10.4 
pandocfilters-1.5.1 
platformdirs-4.2.2 
pyrpl-0.9.6.0 
pywin32-306 
pyzmq-26.0.3 
qtpy-1.9.0 
referencing-0.35.1 
rpds-py-0.18.1 
soupsieve-2.5 
tinycss2-1.3.0 
tornado-6.4.1 
webencodings-0.5.1 
zipp-3.19. 2
```

michaelcroquette, 2024-06-28:
If you create an environment with the right packages, you don't need to run "python setup.py develop". And I think you should not ! This command will use pip as your package manager but you created an environment with conda so you should keep it as your package manager. There can be some issues when the two of them interfere. As for the extra pip packages I think they are dependencies from jupyter probably already included in a default conda environments.

peteasa, 2024-11-01:
I have a potential fix for install problems see - https://github.com/RedPitaya/pyrpl/pull/7/commits/1f1be51364f467f8d9523541cc96a65043b9273c

The objective is to install all the right packages then clone the pyrpl repository .. no need to install pyrpl just link to the pyrpl/pyrpl folder and import the library into your python code from pyrpl folder in your project directory.. I created a demonstration with brief instructions here - https://github.com/peteasa/pyrpl_testbench/blob/main/README.md

closed 2025-10-23 by michaelcroquette