pyrpl-fpga/pyrpl #521: pyrpl-mac is missing

SchneHa, 2024-08-14:
Hi,
I would like to install on my Mac but the file pyrpl-mac is missing. Is there any possibility to install on Mac with ease?

michaelcroquette, 2024-11-28:
https://sourceforge.net/projects/pyrpl/files/

SchneHa, 2024-11-28:
Unfortunately NOT. There is no .dmg or .pkg to install on Mac.

michaelcroquette, 2024-12-02:
I don't have access to a Mac so I can't make a distribution right now. I think the best solution for now is to use the python code.

michaelcroquette, 2026-02-04:
Ok I build the exe on an arm64 mac and you most likely have an intel mac. I will try to fix this.

SchneHa, 2026-02-05:
Same result:
-bash: complete: nosort: invalid option name
/Applications/pyrpl-macos ; exit;
Hans-MBP:~ hasch$ /Applications/pyrpl-macos ; exit;
-bash: /Applications/pyrpl-macos: Bad CPU type in executable
logout

Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.
Deleting expired sessions...none found.

michaelcroquette, 2026-02-08:
Have you tried both files ? 
What do you get when you run `uname -m` in a terminal ? I am not a mac user but it looks like you might need to install rosetta.