pyrpl-fpga/pyrpl #525: "OSError: Socket is closed" Error

sankethan, 2024-10-13:
Im using **redpitaya STEMlab 125-14-Z7020 LN** With OS version **STEMlab_125-14-Z7020_OS_1.04-14_beta** (1.04-14). 

I did the static IP configuration. 

I'm running the python source code installed using anaconda.

I can see my device in the host name selector window.

But when I click on okay I get 

`ERROR:pyrpl.redpitaya_client:Error occured in reading attempt 1. Reconnecting at addr 0x40200008 to _writes value [1073741823] by client 2
WARNING:pyrpl.redpitaya_client:Socket error during connection attempt 0.
WARNING:pyrpl.redpitaya_client:Socket error during connection attempt 1.
WARNING:pyrpl.redpitaya_client:Socket error during connection attempt 2.
WARNING:pyrpl.redpitaya_client:Socket error during connection attempt 3.` 

several times and at the end 

`OSError: Socket is closed`

What am I doing wrong?

michaelcroquette, 2024-11-14:
Are you using the executable or running the python code ?

sankethan, 2024-12-22:
Running The Python Code.