pyrpl-fpga/pyrpl #526: question: does pyrpl support the "slow IO" inputs and outputs of the extension connector?

gsteele13, 2024-10-19:
I could not find it super quickly, so I thought i would ask here

The question is realted to the "slow" analogue IO pins on the extension connector (4 inputs and 4 outputs):

https://redpitaya.com/stemlab-125-14/

I see that the standard API/SCPI server supplied by red pitaya does have commands for interfacing to them:

https://redpitaya.readthedocs.io/en/latest/appsFeatures/remoteControl/command_list.html#analog-inputs-and-outputs

Thanks!
Gary

SamuelDeleglise, 2024-10-19:
It does (at least for the outputs), I believe there are dedicated modules in pyrpl: p.rp.pwm0 and 1 and what you have to do is just set there "input" to whatever signal you want to be sent to them! I am not sure about inputs though

tt-aqt, 2024-10-24:
See issue 476