pyrpl-fpga/pyrpl #528: continuous data acquistion

shivarajkandhasamy, 2024-10-25:
For our project, after locking, we would like to acquire continuous data for a longer time to look at narrow frequency features. Currently, we can only store 16384 samples, which is not sufficient for cause. Is there a way to get more continuous data by continuously capturing the 16384 samples or by some other means? I tried to use the new DMA option in red Pitya (for example [link](https://redpitaya.readthedocs.io/en/latest/appsFeatures/examples/DMA/deepMemoryAcq.html)) along with pyrpl, but they don't seem to work together. If I start the pyRPl and ssh to Red Pitaya to start the DMA acquisition, it doesn't work. But without starting Pyrpl, I can SSH and acquire a longer stream of RF input channels. Let me know if there are options I can explore. Thanks.

michaelcroquette, 2024-11-14:
Have you looked at the sampler module ?

michaelcroquette, 2024-11-21:
Ok I guess I read your question to fast. The sampler module can be used to acquire data for a long time with a low and not well define sampling frequency. 

Regarding your issue, I guess the two ssh connection are clashing. I took a look at your example, are you doing it the same way ? If so an option could be to not open a second connection but to send the SCPI command using the Pyrpl.rp.ssh connection. Not sure if this would work though.