pyrpl-fpga/pyrpl #529: IIR module gain has to be set too small

peteasa, 2024-11-02:
I have been using the IIR module to try and understand how to configure digital filters.  I have a demonstration that is easy to read [here](https://github.com/peteasa/pyrpl_testbench/blob/main/basics/filters.ipynb).  And have been exploring resonance [here](https://github.com/peteasa/pyrpl_testbench/tree/main/filters).  However I seem to have to set the IIR module gain to rather small values to get a working filter.  Being new to pyrpl I would welcome someone who has knowledge of this area of the code to review my findings and give me a hint about how to either debug the IIR module or fix my code!

peteasa, 2024-11-17:
I have been doing more work on this issue.  I have been successful with a notch filter with 
[image]
however with low pass and high pass 
[image]
[image]
[image]
[image]
these all require very low gains for them to work.  I have some demo code for all these filters that I am in the process of analysing but I will publish them soon at https://github.com/peteasa/pyrpl_testbench/tree/main/filters if anyone want to play and provide feedback on this issue!

michaelcroquette, 2024-11-19:
Hello,

I had a quick look at your example. I'm not so surprised at the gain having to be set at a very low value, the transfer function cannot be higher than 0 dB. If I take a gain=1e-8 and loops=10, I can create a decent low-pass filter. But I agree that the behavior of the gain and the loops parameter is quite strange. If I multiply the gain by 10, I would expect the dc gain of the loop to increase by 10 dB and that is not the case. Also the loops parameter changes the dc-gain and I don't think it should. The continuous tf seems to behave normally but the discrete one is shifted by 140 dB so the issue might be here.

peteasa, 2024-12-29:
Having part reviewed the coefficient generation code I suspect that there is a term in dt missing.  The dt term originates from sampling theory when you sample a continuous time system ($h_c$) the result is
$y(nT)\approx\sum^\infty_{k=-\infty}x(kT)h_c((n-k)T) T$
Looking at the function cont2discrete in iir_theory.py this is taken into account, however at some point the code was changes and this function is no longer used. Point 4) in rp2coefficients also comments on this.    

Further inspection shows Point 3) in rp2coefficients states that the partial fraction expansion is performed in the continuous time system.  The code changes have also changed this so now the partial fraction expansion (rd, cd = residues(zd, pd, k)) is done in the z-plane using the z-plane versions of the poles and zeros.

peteasa, 2025-01-06:
Almost completed my review.  I think that the gain question is resolved.  It is working as designed, however I have opened a dsp question about the design of the IIR filter - see https://dsp.stackexchange.com/questions/95906/how-to-debug-this-biquad-iir-filter-that-uses-polyphase-decimation

[image]

michaelcroquette, 2025-01-10:
I tried to respond to your question. If you have a PR correcting the IIR code, it would be very welcomed !

peteasa, 2025-03-14:
closing this as the irr filter is I think working as designed. I may add more to https://dsp.stackexchange.com/questions/95906/how-to-debug-this-biquad-iir-filter-that-uses-polyphase-decimation if I have time!

closed 2025-03-14 by peteasa