pyrpl-fpga/pyrpl #531: GUI Crash

duanmaomao1996, 2024-11-11:
the gui is normal when just write FAKE for IP address, when I entered manual IP address for hostname the exe file suddenly closed by itself. system is windows11 1.04.18

JohannesWes, 2024-11-11:
What error do you get? To get the GUI running, the correct combination of Red Pitaya OS, pyrpl version and python version seems to be important. If you don't require the gui, see the instructions in https://github.com/peteasa/pyrpl/discussions/1 which quite helped me to get pyrpl running with the new RP OS version.

peteasa, 2024-11-17:
interesting problem - I assume that you are using the standard .exe that should work with the older RedPitaya ecosystem.  I am working with the latest eco system and I am using Ubuntu 24.04 where the normal install process does not work and I have a network where I log into multiple machines so the standard pyrpl ssh does not work.  

My fix was to go for the latest of everything, including python3.  In doing that I had as you will see had to do a bit more work but it now works for me!  I am trying to get my code into the main repository however unit testing is not currently working although I am able to run the old unit nose unit tests quite successfully.  The main repository is currently been updated to run pytest rather than nose.  Once that is done things should start to happen!