pyrpl-fpga/pyrpl #533: IQ recorder

nano-aaiv, 2024-11-26:
Hello,

I’m working on an IQ recorder for the Pyrpl FPGA module. The goal is to route the IQ outputs from any of the demodulators to the scope module, allowing for offline processing of the recordings on the PC client. Do you think this is feasible with the current design, or would it be better to consider an alternative?

michaelcroquette, 2024-11-28:
The scope module of PyRPL allows you to record 16000 samples of any signal present inside the board including the IG outputs of the demodulator. The sample frequency goes from 125 MHz down to 2 kHz.

nano-aaiv, 2024-11-28:
Also recording quadrature1 and quadrature2 signals from the iq modules?

michaelcroquette, 2024-12-02:
Yes. I think the section about the Phase Lock Loop developed in this article (https://arxiv.org/abs/2310.00086) could interest you.

nano-aaiv, 2024-12-02:
Thanks! I saw it and they mention a cordic phase estimator that I don't see implemented in any place :( not even at the new_pll branch. The pfd estimation with the rising edges is not sufficent for my application.

michaelcroquette, 2024-12-02:
It is present on the "modif_fpga_pfd" branch and only available for iq2 if I remember correctly.

nano-aaiv, 2024-12-17:
Hey Michael! I found the implementation, fixed some errors and made it work. Do you think would be useful If I post the changes? Or was this like a deprecated functionality that we don't want at the main repo anymore.

michaelcroquette, 2024-12-17:
Hey, Great news ! This branch is still being used and I try to keep it up to date with the main branch. I haven't pulled the most recent changes made so chances are you faced some bug if you used a recent version of python. Anyway if you have some bug fixes, you can post a pull request and I will look at it. Thanks !

nano-aaiv, 2024-12-17:
Cool, I'll post the PR soon as I can. Closing this issue. Thanks for the help!

closed 2024-12-17 by nano-aaiv