pyrpl-fpga/pyrpl #537: how to debug the fpga?

peteasa, 2024-12-09:
I wonder if anyone has used [iverilog](https://steveicarus.github.io/iverilog) to debug the FPGA?  That plus [gtkwave](https://github.com/gtkwave/gtkwave) are a very powerful pair of applications to debug.

It would require a bit of setup with some test scripts but might help debug custom FPGA changes.

peteasa, 2024-12-11:
Here is an example of the output from gtkwave during the startup of the iir module
[image]

nano-aaiv, 2024-12-17:
I haven't! But I'm interested in this topic. What's the advantage of iverilog? Can we use vivado's simulator for this purpose?

peteasa, 2024-12-18:
The advantage of iverilog is that it is quick to make changes and re-run the simulation.  I have pushed my version of the harness here https://github.com/peteasa/pyrpl_testbench/tree/main/fpga.  This does not replace the vivado simulator, its just very quick to run with little overhead.

peteasa, 2025-03-14:
Closing this as I have now got a suitable environment for quickly testing the fpga with iverilog - see  https://github.com/peteasa/pyrpl_testbench/tree/main/fpga

closed 2025-03-14 by peteasa