pyrpl-fpga/pyrpl #552: how to connect pyrpl-windows.exe to my Redpitaya board

jilingcan, 2025-03-24:
I don't know how to connect this Pyrpl-windows.exe to my redpitaya board.
Please show me the basic connection operations.
Thank you very much！

peteasa, 2025-03-28:
have you looked at https://redpitaya.readthedocs.io/en/latest/appsFeatures/applications/pyrpl/pyrpl.html and the associated links?

jilingcan, 2025-04-11:
> have you looked at https://redpitaya.readthedocs.io/en/latest/appsFeatures/applications/pyrpl/pyrpl.html and the associated links?

Yeah!
This pre-built exe file asked me to select the file with the suffix yml, but the window that popped up on my computer did not match the yml file in the official prrpl website.
I think this is the main reason for the inability to connect the development board.
And the IP address window shows nothing!

But this is my first time to use it, so I don't know how to connect it with Redpitaya board.

jilingcan, 2025-04-11:
PyInstaller Bootloader 3.x
LOADER: executable is C:\Users\15135\Downloads\pyrpl-windows.exe
LOADER: homepath is C:\Users\15135\Downloads
LOADER: _MEIPASS2 is NULL
LOADER: archivename is C:\Users\15135\Downloads\pyrpl-windows.exe
LOADER: Extracting binaries
LOADER: Executing self as child
LOADER: set _MEIPASS2 to C:\Users\15135\AppData\Local\Temp\_MEI337322
LOADER: Setting up to run child
LOADER: Creating child process
LOADER: Waiting for child process to finish...
PyInstaller Bootloader 3.x
LOADER: executable is C:\Users\15135\Downloads\pyrpl-windows.exe
LOADER: homepath is C:\Users\15135\Downloads
LOADER: _MEIPASS2 is C:\Users\15135\AppData\Local\Temp\_MEI337322
LOADER: archivename is C:\Users\15135\Downloads\pyrpl-windows.exe
LOADER: SetDllDirectory(C:\Users\15135\AppData\Local\Temp\_MEI337322)
LOADER: Already in the child - running user's code.
LOADER: manifestpath: C:\Users\15135\AppData\Local\Temp\_MEI337322\pyrpl.exe.manifest
LOADER: Activation context created
LOADER: Activation context activated
LOADER: Python library: C:\Users\15135\AppData\Local\Temp\_MEI337322\python34.dll
LOADER: Loaded functions from Python library.
LOADER: Manipulating environment (sys.path, sys.prefix)
LOADER: Pre-init sys.path is C:\Users\15135\AppData\Local\Temp\_MEI337322\base_library.zip;C:\Users\15135\AppData\Local\Temp\_MEI337322
LOADER: sys.prefix is C:\Users\15135\AppData\Local\Temp\_MEI337322
LOADER: Setting runtime options
LOADER: Bootloader option: pyi-windows-manifest-filename pyrpl.exe.manifest
LOADER: Initializing python
LOADER: Overriding Python's sys.path
LOADER: Post-init sys.path is C:\Users\15135\AppData\Local\Temp\_MEI337322\base_library.zip;C:\Users\15135\AppData\Local\Temp\_MEI337322
LOADER: Setting sys.argv
LOADER: setting sys._MEIPASS
LOADER: importing modules from CArchive
LOADER: extracted struct
LOADER: callfunction returned...
LOADER: extracted pyimod01_os_path
LOADER: callfunction returned...
LOADER: extracted pyimod02_archive
LOADER: callfunction returned...
LOADER: extracted pyimod03_importers
LOADER: callfunction returned...
LOADER: Installing PYZ archive with Python modules.
LOADER: PYZ archive: out00-PYZ.pyz
LOADER: Running pyiboot01_bootstrap.py
LOADER: Running pyi_rth_qt5.py
LOADER: Running pyi_rth_qt5plugins.py
LOADER: Running pyi_rth_pkgres.py
LOADER: Running pyi_rth_win32comgenpy.py
LOADER: Running __main__.py
0 C:\Users\15135\Downloads\pyrpl-windows.exe
Calling Pyrpl(**{})
INFO:pyrpl:Please select or create a configuration file in the file selector window!
INFO:pyrpl.redpitaya:Please choose the hostname of your Red Pitaya in the hostname selector window!
WARNING:pyrpl.modules:Register decimation of module scope has value 1, which does not correspond to selected option 8192. Setting to '8192'.
INFO:pyrpl.redpitaya:Successfully connected to Redpitaya with hostname 192.168.8.191.
Traceback (most recent call last):
  File "pyrpl\__main__.py", line 37, in <module>
  File "pyrpl\pyrpl.py", line 290, in __init__
  File "pyrpl\software_modules\network_analyzer.py", line 348, in _load_setup_attributes
  File "pyrpl\modules.py", line 484, in _load_setup_attributes
  File "pyrpl\modules.py", line 471, in setup_attributes
  File "pyrpl\modules.py", line 158, in setup
  File "pyrpl\software_modules\network_analyzer.py", line 662, in _setup
  File "pyrpl\software_modules\network_analyzer.py", line 629, in _update_data_x
  File "pyrpl\software_modules\network_analyzer.py", line 364, in iq
  File "pyrpl\attributes.py", line 240, in __set__
  File "pyrpl\attributes.py", line 720, in validate_and_normalize
  File "pyrpl\attributes.py", line 720, in <listcomp>
  File "pyrpl\attributes.py", line 793, in valid_frequencies
  File "pyrpl\attributes.py", line 785, in _MAXSHIFT
ZeroDivisionError: float division by zero
Failed to execute script __main__
LOADER: OK.
LOADER: Cleaning up Python interpreter.

I looked at the link.
And why Failed to execute script __main__.