pyrpl-fpga/pyrpl #555: Decoupling of PIDs

gtomaras18, 2025-04-15:
Greetings pyrpl team,

We want to stabilize a laser beam at two points (quadrant photodiodes) with two piezo-controlled mirrors (each mirror has vertical and horizontal tilt dofs), stabilized to each photodiode with a PID loop. As expected, the two PIDs couple. To decouple them, we start by determining the transfer matrix that quantifies the effect of each of the two dofs of each of the two mirrors on the outcome of each of the two outcomes from each photodiode. Then, we compute the inverse matrix and find the corrections that need to be applied to each piezo to remove its effect on the photodiode signal (decoupling).

The question: How to go about implementing this calculation on the outputs of the PIDs in pyrpl? The easiest for me would be to implement this calculation in the .yml configuration file, since we work with the GUI.
Any advice is appreciated.

Thank you for your help, and be well.

George

michaelcroquette, 2025-04-17:
Hello,

Do you just want to use a linear combination of your error signals or do you want to do a more complex treatment ? You can just use two PID modules outputting on the same channel and it should do the trick but you would need four PID modules in total.  The 'modif_fpga_pfd' branch has 4 PIDs so you can give it a try (might need to merge recent commits from main).

gtomaras18, 2025-04-21:
I want to use a linear combination of error signals. I will try what you suggest.

Thank you and be well.

closed 2025-07-01 by michaelcroquette