pyrpl-fpga/pyrpl #559: Documentation for 0.9.6 ?

brian-sinquin, 2025-05-21:
Hello,

Is there somewhere an up to date documentation with the last API ? (version 0.9.6 Python3)

peteasa, 2025-05-21:
The code is well structured and has documentation in it and the examples also have documentation that then auto generates some or all of the https://pyrpl.readthedocs.io/en/latest/index.html pages.  So the readthedocs pages are a good point of reference.  As you will spot the documentation has not kept up to date with the code.  Here are my notes and example code using the python3 branch - https://github.com/peteasa/pyrpl_testbench/tree/main/basics#tips-and-tricks. Not much I agree but the job of updating the documents properly is not trivial and will take time and energy!

brian-sinquin, 2025-05-21:
> The code is well structured and has documentation in it and the examples also have documentation that then auto generates some or all of the https://pyrpl.readthedocs.io/en/latest/index.html pages. [...]

Thank you very much !