pyrpl-fpga/pyrpl #564: Rotary encoders ?

jl0up, 2025-06-30:
Hi!
So, pyrpl on a Red Pitaya makes an awesome PID and knobs are great for PIDs, and Red Pitaya has GPIOs that I believe can polled fast enough to read a rotary encoder.
I believe changes to the FPGA could be made so that the P register (interfaced by `pyrpl.rp.pid.p`) has a value modified in real-time as the encoder rotates. But it's a bit complicated to modify the FPGA.
If a slow (not real-time) update is enough, is there a software way to have P (or I, or any scalar) changed using the encoder? Can the computer running Pyrpl define a callback than adjusts pid.p upon a change on GPIOs ?

michaelcroquette, 2025-07-01:
I guess you could code a async python routine that would read the value of the GPIOs and link it to the value of  pyrpl.rp.pid.p. The upadte time would be of the order of 10 ms depending on your network connection to the RP.

jl0up, 2025-07-04:
Thanks, I will try