pyrpl-fpga/pyrpl #568: Scope measures voltage wrongly

wwlyn, 2025-07-18:
Scope cannot measure voltage correctly. I send a 0.5V dc to the in1, but the scope said it is 0.003V (It does change compared to no signal input). Because of this, I cannot use PID module.

peteasa, 2025-07-22:
The pyrpl code does not have access to the calibration values stored by the normal RedPitaya calibration application.

As a result you need to think about calibration for the specific applications that you have.  I discuss this in my [pyrpl_testbench](https://github.com/peteasa/pyrpl_testbench?tab=readme-ov-file#pyrpl-calibration) and have been experimenting with various ways to calibrate the various tools.

I also have a demonstration of the PID module in the same repository that may or may not help with your work!

wwlyn, 2025-07-23:
> The pyrpl code does not have access to the calibration values stored by the normal RedPitaya calibration application.
>
> As a result you need to think about calibration for the specific applications [...]

Yes! My first big offset is beacause I set the HV mode but forgot to add gain! 
But in LV mode, the offset still ramains. I look deep into the web calibration result and it's correct. But in the scope module of pyrpl, it's wrong! It seems you are right!

[image]

[image]

I'll try to fix it, thanks for your help!! And I'll tell you the result.

closed 2025-10-23 by michaelcroquette