pyrpl-fpga/pyrpl #571: which branch is the IIR on?

Amany-k, 2025-07-27:
Hello,

I'm currently working with the modif_fpga_pfd branch to lock two lasers using the pll example. I've successfully achieved locking using a two-stage sequence with the fast and slow piezos, and I've measured the loop transfer functions. I'm now looking to improve the response of the fast piezo by adding a series IIR filter.

To get familiar with the IIR filter, I tried to experiment with it separately. However, the module doesn’t seem to be available in this branch. When I simply try:

iir = p.rp.iir

I get the following error:

AttributeError: 'RedPitaya' object has no attribute 'iir'

- Is the IIR module not included in the modif_fpga_pfd branch?

- If not, which branch should I use to access the IIR filter while retaining support for cordicerror, which seems to only be working on modif_fpga_pfd?

michaelcroquette, 2025-07-29:
Hello,

Good to know this branch is being used ! I'm currently working on it to make the cordic module more user-friendly so be careful if you pull the most recent commits. 

So the IIR module has been removed from this branch because it requires a lot of FPGA ressources. It was a long time ago so I could try to rebuild it now but I think it would fail. Which hardware modules are you using ? I guess at least one PID and one IQ module. We could try to build an FPGA image with less PID and IQ modules and see if we can make the IIR filter and CORDIC algorithm work together on one board.

michaelcroquette, 2025-09-03:
> I have successfully locked the lasers, tuned p and i factors and measured the phase noise if the locked lasers, which is not as good as expected (I have performed analog locking with these before and it was better), which brings me to think that the performance can be improved using Pyrpl.

Cool ! Glad someone is using this feature. You should be able to get a lock as good as you had with the analog device. 

> In practice, I’ve noticed that the second stage (slow piezo) doesn’t improve the phase noise performance, it mainly just re-centers the fast piezo range because it acts on the errorfastpiezo, not the CORDIC error. 

Ok so what controls do you have on your laser ? In our setup we have a fast piezo that controls the laser over a very small range. This output is directly connected from the RedPitaya output to the piezo of the laser without any filtering or amplifier. So it's a very fine output with ony some small digital noise coming from the card. The slow piezo output is amplified (~x50 if I remember correctly) and low pass filtered. So it has a much bigger tuning range to compensate slow drift. The slow piezo and fast piezo outputs are respectively connected to the "+" and "-" terminals of the laser piezo. The slow stage is to compensate the even slower and bigger drift as it has a much bigger tuning range and even slower bandwidth. 

>So in effect, the locking is only as good as the first (fast) stage, so if we then add an IIR to the fast piezo stage wouldn't that improve the locking? 

The IIR is not magical. It will help if what limit your lock is a resonance of your system. To answer this question, I advise you to look at the example notebooks 5 and 6 from the article in docs/example-notebooks/article_examples. You should also check the Pyrpl [article](https://hal.science/hal-04800399v1) for more informations if you haven't. 

>Also if left running for a while, the system sometimes briefly unlocks; my guess is that because I’m not using the temperature stage, nothing re-centers the slow piezo, so over time it can drift and compromise stability.

This make sense if you have no slow output with a large tuning range to recenter your error signal. What happens when it unlocks ? Do the fast piezo saturate to +1 or -1 V ? If so then you need an output with a larger tuning range to compensate for the slow drifts.
  
> I’m relatively new to digital locking, but I’ve performed similar locking with analog electronics before, also using two stages both acting on the initial error. In the analog case, once the first stage was locked, it generally stayed locked without needing to recenter the error and since the second stage also acted on the initial error it led to an improvement in the phase noise performance.

What was the difference betweent the two stages then ?

> 
> So my question is:
> 
> * What’s the rationale for the re-centering strategy in the digital lockbox (using errorfastpiezo for the slow stage) rather than having both loops act on the same CORDIC error with different bandwidths? when I tried having the two stages acting on the CORDIC error, I was able to see further improvement in phase noise but it didn't stay locked for long.

To compensate slow drifts. If you use the CORDIC error for you slow ouput it won't do anything because its average value is maintained around zero by the fast piezo. The fast piezo average value will drift slowly and you can compensate that with the slow piezo.

> * When the fast loop is “locked,” what causes the offset to drift until the actuator saturates? is there a way to solve this issue without having a second stage recentering the first?

Well the overall frequency stability of your two lasers. So it will depend on your setup basically.

> * Is temperature control necessary in this architecture to re-center the slow piezo over long timescales, or can we achieve robust locking with just two piezo loops (fast + slow) and appropriate filtering.

Again this will depend on the overall stability of your setup. If you see that the slow piezo saturates after some time then you will know.
> 
> Finally, if it’s feasible to compile an FPGA image with IIR + CORDIC while maintaining one IQ module for the CORDIC error with the fast and slow piezo (so I guess in that case two PID modules), I’d be very interested in trying that approach and checking if it makes a difference to the phase noise.

I would have to try and this would take a bit of effort. So I think you should make sure that the IIR will actually be beneficial for you. The best way is to plot the loop transfer function of your system as explained in example 5, but a fast test you can do is to increase your gain and see what happens. If the system stays locked but starts to oscillate than most likely you are indeed limited by a resonance of your setup and the IIR could help. You can use the spectrum anlyzer to look at the error signal and estimate the frequency of this resonance. Pyrpl is a great tool for controlling your system but it is even better at diagnosing it !