pyrpl-fpga/pyrpl #572: How can I change maximum integral gain constant?

wwlyn, 2025-07-30:
The maximum integral gain is 3.8e4, where does it comes from? Is it from the limitation of redpitaya's hardware or just the software? How can I set it to be higher?

wwlyn, 2025-07-30:
> If you look in pyrpl\hardware_modules\pid.py, you can find the definition of the integral gain:
> 
> ```
> _delay = 3  # min delay in cycles from input to output_signal of the module
>     # with integrator and derivative gain, delay is rather 4 cycles
> 
>     _PSR = 12  # Register(0x200)
> 
>     _ISR = 32  # Register(0x204)
> 
>     _DSR = 10  # Register(0x208)
> 
>     _GAINBITS = 24  # Register(0x20C)
> 
>     ival = IValAttribute(min=-4, max=4, increment=8. / 2**16, doc="Current "
>             "value of the integrator memory (i.e. pid output voltage offset)")
> 
>     setpoint = FloatRegister(0x104, bits=14, norm=2**13,
>                              doc="pid setpoint [volts]")
> 
>     min_voltage = FloatRegister(0x124, bits=14, norm=2**13,
>                                 doc="minimum output signal [volts]")
>     max_voltage = FloatRegister(0x128, bits=14, norm=2**13,
>                                 doc="maximum output signal [volts]")
> 
>     p = GainRegister(0x108, bits=_GAINBITS, norm=2**_PSR,
>                       doc="pid proportional gain [1]")
>     i = GainRegister(0x10C, bits=_GAINBITS, norm=2**_ISR*2.0 * np.pi *
>                                                   8e-9,
>                       doc="pid integral unity-gain frequency [Hz]")
> ```
> 
> If you increase ISR you should be able to reach higher value but you will reach saturation quicker. You would need to change the value in pyrpl\fpga\rtl\red_pitaya_pid_block.v and rebuild the FPGA.

Thank you so much! I'm looking into it. My concern is that there should be a reason for pyrpl authors to set such a limitation, like the limitation of the hardward.... Anyway, I'll try to do that. Thank you so much!

michaelcroquette, 2025-07-31:
If you don't increase GAINBITS the complexity of the calculations should stay the same, you are just giving the integrator more "weight". Anyway you will see when you build the FPGA.

closed 2025-10-23 by michaelcroquette