pyrpl-fpga/pyrpl #573: How can I change maximum integral value?

wwlyn, 2025-07-30:
The maximum integral value is +4V, and the minimum is -4V. How can I change it to +1V and -1V?

michaelcroquette, 2025-07-30:
If you look at my previous comment you will see the definition of ival is:

```
     ival = IValAttribute(min=-4, max=4, increment=8. / 2**16, doc="Current "
            "value of the integrator memory (i.e. pid output voltage offset)")
```

So you can change -4 and 4 to -1 and 1 and 8. to 2. but this will have no effect on the actual value of the register in the FPGA code.

wwlyn, 2025-07-30:
> If you look at my previous comment you will see the definition of ival is:
>
> ```
>      ival = IValAttribute(min=-4, max=4, increment=8. / 2**16, doc="Current "
>             "value of the integrator [...]

Yes! I'm trying to change the FPGA code. I want to do that because I want to get fastest respond of pid. If there is a gap between ival and the real output, sometimes, pid needs some time to do the -4V to -1V to have a response.