pyrpl-fpga/pyrpl #576: Calibration error：asyncio.exceptions.InvalidStateError: Result is not set.

Thomas428-jjg, 2025-08-26:
Hello, I'm sorry to disturb you, but I have a problem that has been persistently unresolved. I'm currently using Pyrpl, and this tool is extremely powerful. I created the environment in Anaconda using the "environment_pyrpl.yml" file, and I was also able to successfully connect to my Redpitaya. However, every time I click the calibration button, it always gives an error. I have no idea what to do. Could you offer some suggestions? Here is my work log.

(pyrpl_env_2) E:\pyrpl-main\pyrpl-main>python
Python 3.9.19 (main, May  6 2024, 20:12:36) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from pyrpl import Pyrpl
C:\Users\15721\.conda\envs\pyrpl_env_2\lib\site-packages\paramiko\transport.py:219: CryptographyDeprecationWarning: Blowfish has been deprecated and will be removed in a future release
  "class": algorithms.Blowfish,
>>> p = Pyrpl("try6")
INFO:pyrpl:All your PyRPL settings will be saved to the config file
    C:\Users\15721\pyrpl_user_dir\config\try6.yml
If you would like to restart PyRPL with these settings, type "pyrpl.exe try6" in a windows terminal or
    from pyrpl import Pyrpl
    p = Pyrpl('try6')
in a python terminal.
INFO:pyrpl.redpitaya:Please choose the hostname of your Red Pitaya in the hostname selector window!
WARNING:pyrpl.redpitaya:no pyrpl instance
WARNING:pyrpl.redpitaya:no pyrpl instance
INFO:pyrpl.redpitaya:Successfully connected to Redpitaya with hostname 169.254.44.240.
>>> WARNING:pyrpl.modules:Register input of module pwm0 has value 15, which does not correspond to selected option 10. Setting to 'in1'.
QLayout::addChildLayout: layout "" already has a parent
E:\pyrpl-main\pyrpl-main\pyrpl\software_modules\lockbox\models\fabryperot.py:115: RuntimeWarning: invalid value encountered in true_divide
  return(pdh(x)/pdh(x_max))
E:\pyrpl-main\pyrpl-main\pyrpl\software_modules\lockbox\models\fabryperot.py:115: RuntimeWarning: invalid value encountered in double_scalars
  return(pdh(x)/pdh(x_max))
Traceback (most recent call last):
  File "E:\pyrpl-main\pyrpl-main\pyrpl\widgets\module_widgets\lockbox_widget.py", line 493, in <lambda>
    self.button_calibrate.clicked.connect(lambda: self.module.calibrate())
  File "E:\pyrpl-main\pyrpl-main\pyrpl\software_modules\lockbox\input.py", line 309, in calibrate
    curve, times = self.sweep_acquire()
  File "E:\pyrpl-main\pyrpl-main\pyrpl\software_modules\lockbox\input.py", line 294, in sweep_acquire
    curve1, curve2 = scope.single(timeout=1. /
  File "E:\pyrpl-main\pyrpl-main\pyrpl\acquisition_module.py", line 294, in single
    return wait(self._renew_run(self._single_async()), timeout=timeout)
  File "E:\pyrpl-main\pyrpl-main\pyrpl\async_utils.py", line 164, in wait
    done, pending = new_future.result()
asyncio.exceptions.InvalidStateError: Result is not set.
If you can help me, I would be extremely grateful.

Thomas428-jjg, 2025-08-26:
I am using Redpitaya 0S1.04-18/STEMlab 125-14 and I cloned the latest pyrpl from the website.

michaelcroquette, 2025-09-02:
I was able to reproduce the bug. If you run it from jupyter for now it should work. You can also try a newer version of python.

Thomas428-jjg, 2025-09-03:
Thank you very much for your suggestion. By running it from Jupyter, this error was indeed resolved.